LabVIEWCommunityTraining/GettingStartedLabVIEW1-English #1: Multinlingual documentation

OlivierJourdan, 2024-05-01:
Hey there,
I just listened to the first episode of TImLV and heard @swatzy talk about translating the documentation into different languages.
I see that there are two different repos for English and Spanish. Content is written with libre office.

I'd like to propose a different way to deal with the documentation (translation and evolution) to have a more efficient (in my humble opinion) way to handle the documentation.

The idea is to have only one repo with content written using the asciidoc syntax. This will allow all languages in the same place (using the same image resources) and generate PDFs for each language automatically.

Let me know if you are interested. I could try to find some time to work on a proof of concept to see if it is worth the effort.

j-medland, 2024-10-24:
I am all for standardizing on a text-based format.

I tried converting [some of the docs to markdown with mkdocs for rendering](https://j-medland.github.io/test_cti_docs/Getting-Started-LabVIEW-1/Course-Manual/) which seemed OK although markdown doesn't provide much image sizing and positioning without HTML hacks. I also liked having admonitions.

The presence of lack of these features in asciidoc (and whatever rendering process we use) isn't a deal breaker, I am just offering my two ¢.

unifab, 2024-10-24:
@OlivierJourdan I was trying to find out where you had offered to help and I am so glad to finally find it! (@j-medland thanks for commenting and causing this to pop into my radar).

I am working on organizing the translation of the CTI material. I agree with the one-repo approach and would love to discuss ways around the issues John is mentioning.  I know that for the DQMH LabVIEW Style Guidelines we figured out ways to highlight things in colors and have Warnings (or as John called them, admonitions). I will be pinging you offline, maybe we can discuss this next time we meet and then I will bring back the resolution to this case and the Discord channel discussion.

Thanks,
Fab

unifab, 2024-10-25:
@j-medland,

Would something like the pin icon on this document would be enough for the "admonitions"? 

https://gitlab.com/dqmhconsortium/open-source/development-guidelines

j-medland, 2024-10-27:
Hey @unifab 

I don't really mind. Having some option to break up long sections of text, especially if you may or may not be using real hardware is handy.

I am happy to work with whatever consensus is reached on what features we use in the documentation.  

I would assume that the raw markdown/ascii doc files would be built into a static HTML site so we aren't relying on the features only supported by the GitHub adhoc page rendering.

OlivierJourdan, 2024-10-28:
I like the idea of the static HTML site independent of the DevOps platform used.

enoearias, 2024-10-28:
Hi, I´m using Github Pages for the Spanish repo, it renders the markdown on every commit so I can see if the changes are what I was expecting, what is the difference between this and static HTML?

j-medland, 2024-10-28:
@enoearias - There is no difference - your source files are built to static HTML. The build does the magic of converting markdown to html and applies styling/CSS based on the theme you have chosen.

I guess I just wanted to distinguish that we would not rely on whatever github gives us in the repository browser i.e.

https://github.com/LabVIEWCommunityTraining/GettingStartedLabVIEW1-Spanish/blob/main/docs/InstruccionesTutor.md

But would instead build the pages into html and then direct users to the static HTML site i.e.

https://labviewcommunitytraining.github.io/GettingStartedLabVIEW1-Spanish/InstruccionesTutor.html

The choice of what formatting we can use in the documentation is restricted by what the static-site generator will support but hopefully it should be better than the basic Github rendering in the repository browser.

unifab, 2024-10-28:
@enoearias,
It's the same thing. You are using Markdown and Jekyll, while we are thinking of using Asciidoc and Antora.
@OlivierJourdan proposes Asciidoc because it has a clear standard syntax whereas Markdown has several versions. Also, Asciidoc does support UTF8 for the paragraph content, the syntax is ASCII only.
Olivier will research how best to support multiple languages on the final HTML page.

What we are envisioning at this time is:

- Have one repository dedicated to the documentation
- Each document would support all the languages, we can see the paragraph in all the different languages.... for someone like me who speaks multiple languages, I may be able to update several languages at a time on the same document.
- We will use syntax for the Automatic Action to decide which paragraph to use for each HTML page.
- We will use GitHub pages to publish the documentation.
- GCentral will link to those HTML pages.

j-medland, 2024-10-30:
Just a thought here on using one document per section with multiple languages - this is a bit different to how many other multilingual docs are constructed where different languages live in parallel sites.

I think the combined document approach optimizes for small changes and fixes but larger changes have to happen all at once and in multiple languages so blocks and their translations stay in sync.

I am happy to try and contribute either way, I am just throwing this idea into the mix for discussion. It might be entirely moot, especially as large section-level changes are probably not that common once the documentation outline for a topic/course has been fixed.

OlivierJourdan, 2024-10-30:
You're right @j-medland. I couldn't find (didn't spend that much time, to be fair) examples where all languages are in the same files. There are probably reasons for that…
Nice to have your help on that topic.

@unifab, what are the arguments that speak against trying the one file per language approach?

enoearias, 2024-11-01:
I would like to have all the languages separated and Add  the appropriate links to navigate to any other language from each of the sections.
For example, if the user is reading my Tutor instructions, at the top of this page add a way to navigate to the corresponding Tutor Instructions in all the other available languages.

I think we all should only agree on the sections, I would follow these as the main categories / sections for all the languages:

- Tutor Instructions 
- Course Manual 
- LabVIEW Instrument Emulator
- LabVIEW Instruments Driver 
- Examples
- Solutions
- Firmware

I've noticed in the English repo the tutor instructions is the second element and I feel this should be the first one, if you don´t follow the tutor instructions, you can't do the Course manual, also maybe, the tutor instructions should be just **Instructions**, there will be a lot of people trying this by themselves.

The Examples, Solutions, Firmware will live only in the English repository, all the other languages will just point to the corresponding section in the English repo.
For the Examples, Solutions, maybe we all could work togheter to add notes to the block diagrams in different languages, for the Spanish community I don't think this is a must but I can happily do it if we all agree this is a good idea.

enoearias, 2024-11-01:
Additionally, all the images can live in the English Repo and all the other languages just points to the Image URL to keep everything in one single place, what I did in the Spanish repo was just dragging the image to the Github markdown editor and it creates the Markdown image reference for me, it was easier but if you all prefer a single source of truth regarding the images I´ll change the urls, it is an easy task, and I can see the advantage of doing  that.

unifab, 2025-02-26:
@swatzy 

We have the first pass ready for you to review for the documentation in English: https://labviewcommunitytraining.github.io/cti-documentation/en/cti-doc/index.html

There are things that I need to change like removing the instructions to log in as root since the new version of the Virtual Machine won't need that... but before I go on making changes, I want to make sure we have a single source for the Online Documentation.

I am going to assign this case to you. Would you be able to assign it back to me with your approval to remove the odt files from the GettingStartedLabVIEW1-English repository? I will make sure the README and the repo itself point on the right of the page to https://labviewcommunitytraining.github.io/cti-documentation/en/cti-doc/index.html

Thanks,
Fab

swatzy, 2025-02-26:
Go for it, what else do I need to do?

unifab, 2025-02-26:
@swatzy 

That is all I needed. I went ahead and removed the odt files and edited the README file to reflect where the documents are now. I also updated the web view of the repo to point to https://labviewcommunitytraining.github.io/cti-documentation/en/cti-doc/index.html

I documented the commits as well as I could. I left the PDN files and the software in this repository.

This continues to have all the software.

closed 2025-02-26 by unifab