LabVIEWCommunityTraining/GettingStartedLabVIEW1-English #7: Move LabVIEW code out of instr.lib

crossrulz, 2024-09-20:
The driver VIs are currently set to be in instr.lib.  They should be saved to link outside of instr.lib for the sake of the repository.

unifab, 2025-03-02:
@crossrulz,

Having them in instr.lib means that people can drag them into the instr.lib and have the same experience as downloading a Plug and Play instrument driver. I am about to make a change so the example shows up in the Example Finder, and I found this case from you.

Unfortunately, NI expects you to develop the instrument driver under instr.lib where it will be installed.

What would be the alternative that you propose that would still allow the plug and play instrument driver experience?

Thanks,
Fab

crossrulz, 2025-03-02:
Eventually, a VI Package will be created that can be installed into instr.lib and Examples. For the sake of the repo, it just made more sense to develop with the rest of the repository. This is not something I'm going to push on for this repo as I have plans to move the course drivers into a different repo once the cti-drivers-lv-visa has another release.

unifab, 2025-03-02:
Sounds good, I am assigning this case to you. 

The example bin file is now in there, so whenever you create the package, please make sure to include it so we don't lose the ability to find the example via the Example Finder.

Thanks,
Fab