LabVIEWCommunityTraining/GettingStartedLabVIEW1-English #11: SQLite Error LabVIEW Component is missing (LCOD)

unifab, 2025-02-22:
Left a note inside the ErrorLogDialog.vi inside the Emulator project for Steve to fix: 

" #CodeReview-To-Do [Fabiola De la Cueva - Feb-22-2025] This VI is missing dependencies and it does not seem to be called by the rest of the code. 
I am disabling it so I can save the rest of the VIs for LabVIEW 2020 and already reached out to Steve to see what he 
wants me to do with this VI." 

The SQLite component seems to be missing.

Steve, when you work on this, please make sure you are working and save for LabVIEW 2020.

swatzy, 2025-02-28:
The error dialog is not required (I do all the error stuff as part of the emulator display). The fix is just to remove ErrorLogDialog from the project.
I will attempt to do this today.. it's good learning for me.

unifab, 2025-03-02:
@swatzy,

Error is gone and I have verified that both the develop and the main branch have the change.

I am closing this issue as solved.

Thanks,
Fab

closed 2025-03-02 by unifab