LabVIEWCommunityTraining/GettingStartedLabVIEW1-English #14: Modify the HandsOn Pi2040 project so the example is found via the Example Finder

unifab, 2025-03-02:
To improve the experience when dragging the HandsOnPi2040 into instrlib and have it behave as any other instrument driver, we need to have the example VI outside of the library in an Example folder and prepare the example so the Example Finder finds it.

unifab, 2025-03-02:
@enoearias,

Would you please verify that if you download the latest version of the code and drag HandsOnPi2040 to instr.lib, next time you open LabVIEW, you can see the Pico example under the Instruments folder?

Also, you should be able to find it by searching for keywords: Pi2040, Pico, or CTI

If it works on your end, please close this case as solved; if not, please assign it back to me with steps to reproduce so I can fix it.

Thanks,
Fab

enoearias, 2025-03-02:
Hi @unifab, everything worked as extpected, closing this issue.

For the record, I struggled finding Instr.lib paths under  Linux, so to not forget and for anyone trying to do the same these are the paths:

- Linux:

/usr/local/natinst/<LabVIEW-VERSION>/instr.lib

- Windows:

C:\Program Files (x86)\National Instruments\<LabVIEW-VERSION>\instr.lib

Where LabVIEW-VERSION is the installed one.

closed 2025-03-02 by enoearias

unifab, 2025-03-02:
@enoearias 

I was struggling too, and then I went to check this section of the documentation and got the path from there: https://labviewcommunitytraining.github.io/cti-documentation/en/cti-doc/latest/index.html#installing-drivers