LabVIEWCommunityTraining/GettingStartedLabVIEW1-English #16: No longer need to support Jekyl GitHub Pages for this repo

unifab, 2025-03-04:
@enoearias,

I believe you had set the Getting Started English to publish using Jekyl, correct? 

Since we are no longer using that and we are using the CTI-documentation URL, are you OK with us removing that? We are getting failures when we commit and build.

I believe the only thing that was building was the GitHub page. Can you confirm that?

Thanks,
Fab

enoearias, 2025-03-04:
Hi @unifab , yes, I enabled that only for the GitHub Page. I don't mind if that gets removed.

unifab, 2025-03-04:
@enoearias,

Please remove the support for GitHub Pages.

Thanks,
Fab

enoearias, 2025-03-04:
@unifab, Jekyl Workflow removed, please let me know if that solved the related issues.

closed 2025-03-05 by unifab